pololu/a-star #5: Question: using the Arduino Ethernet shield on top of the A-Star 32U4 Prime SV

crazyquark, 2018-06-08:
Hello,

I am trying to build an SD card cloning/recovery thing and I want to use both the included sd card on the A-Star 32U4 board and the one on the Arduino Ethernet shield, the problem is once I connect the shield, no matter what I change the chip select pin to it only connects to the one on the shield.
Are they both using pin 4 by any chance for CS?

kevin-pololu, 2018-06-08:
Hi,

The SD card connector's CS pin is not connected by default, so you need to connect it to the pin of your choice and make your code match. Please see the "microSD card connector and level shifters" heading in the ["A-Star 32U4 Prime pinout and components" section of the user's guide](https://www.pololu.com/docs/0J61).

If you continue to have trouble, please post a question on the [Pololu forum](https://forum.pololu.com/), as the GitHub issues are intended for keeping track of problems with and changes to the library itself.

Kevin

closed 2018-06-08 by kevin-pololu

crazyquark, 2018-06-09:
Sorry, thanks for this