pololu/a-star #6: I have locked my atmega328PB when i try to use the bootloader

ZoomStar, 2018-11-22:
Hi... 

I have followed the tutorial which describe how to install the bootloader over an atmega328PB..

But when i tried to to that...the installation showed "Bootloader failed" but...now the cip is not detectable...is just stop working ...and i think the bootloader has reseted the fuse or something....

This is for an important project for my electrical engeennering course...and i don;t want to fail...please help me with this isseu...

Here is what arduino ide displayed when i tried to burn the bootloader...

`Arduino: 1.8.5 (Windows 10), Board: "Pololu A-Star 328PB, 5V, 16MHz"

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -C+C:\Users\georg\AppData\Local\Arduino15\packages\pololu-a-star\hardware\avr\4.0.2/extra_avrdude.conf -v -patmega328pb -cusbasp -Pusb -B 5 -e -Ulock:w:0xFF:m -Uefuse:w:0xF4:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"
         Additional configuration file is "C:\Users\georg\AppData\Local\Arduino15\packages\pololu-a-star\hardware\avr\4.0.2/extra_avrdude.conf"
avrdude: warning at C:\Users\georg\AppData\Local\Arduino15\packages\pololu-a-star\hardware\avr\4.0.2/extra_avrdude.conf:1: part m328pb overwrites previous definition C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf:8710.

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 5.0
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 187500 Hz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip
avrdude: set SCK frequency to 187500 Hz
avrdude: reading input file "0xFF"
avrdude: writing lock (1 bytes):

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -C+C:\Users\georg\AppData\Local\Arduino15\packages\pololu-a-star\hardware\avr\4.0.2/extra_avrdude.conf -v -patmega328pb -cusbasp -Pusb -B 0.5 -Uflash:w:C:\Users\georg\AppData\Local\Arduino15\packages\pololu-a-star\hardware\avr\4.0.2/bootloaders/optiboot/optiboot_atmega328pb_16mhz.hex:i -Ulock:w:0xCF:m 
Writing |  ***failed;  
[... 97 lines cut ...]
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 1500000 Hz
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
	`

Thank you in advance

per1234, 2018-11-24:
Related report: https://github.com/MCUdude/MiniCore/issues/73 (note the hypothesis here: https://github.com/MCUdude/MiniCore/issues/73#issuecomment-441310492).

DavidEGrayson, 2018-11-24:
Loading invalid clock settings onto an AVR can render it unprogrammable.  The bootloader you are trying to load onto your ATmega328PB requires that it have a 16 MHz crystal/oscillator connected to it.

You could just buy the [Pololu A-Star 328PB 5V, 16 MHz](https://www.pololu.com/product/3160) and use that.  It already has a bootloader on it.

Or you could try getting the [Pololu USB AVR Programmer v2.1](https://www.pololu.com/product/3172) and following the instructions in the "Using the clock output to revive AVRs" section of its [user's guide](https://www.pololu.com/docs/0J67) to repair your ATmega328PB.

closed 2019-11-25 by DavidEGrayson