pololu/a-star #7: Error while uploading Arduino sketch on Mac OS Catalina (10.15.1)

bdelapen, 2019-11-25:
Hello,

With its newest update Mac OS Catalina has introduced a serial port error into the Arduino IDE when uploading sketches to boards with the ATmega32U4 processor (Leonardo Architecture). Details on the error can be found here:

https://github.com/arduino/Arduino/issues/9290

One of the suggested workarounds while we wait for Apple to potentially revert to the old USB behavior is to modify the bootloader on the boards. It seems that a small change to the bootloader will allow the boards to upload sketches again. The specific details of the change are shown here:

https://github.com/arduino/Arduino/issues/9290#issuecomment-542728021

I have attempted to set up an environment to compile an updated bootloader but I can only assume that I am doing something wrong as my bootloader does compile and I can burn it to the board with a USBASP adapter but when I try to upload the sketch I get a different error.

I was wondering if it would be possible to get an "official" updated version of the bootloader that I could burn on to the board and test (in case I am not compiling my version correctly)?

Thanks,

- Birger

For reference, below are the errors that I am getting (verbose version at the bottom).
Original bootloader error:

```
Arduino: 1.8.10 (Mac OS X), Board: "Pololu A-Star 32U4"

Sketch uses 3966 bytes (13%) of program storage space. Maximum is 28672 bytes.
Global variables use 149 bytes (5%) of dynamic memory, leaving 2411 bytes for local variables. Maximum is 2560 bytes.

avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_drain(): read error: Device not configured
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = ""; type =  
    Software Version = . ; Hardware Version =  . 
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
avrdude: ser_close(): can't reset attributes for device: Device not configured
the selected serial port avrdude: ser_close(): can't reset attributes for device: Device not configured
 does not exist or your board is not connected
```

"Patched" bootloader error (possibly incorrectly patched?), this error takes a lot longer to generate, possibly due to multiple timeouts:

[... 121 lines cut ...]
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
the selected serial port avrdude: initialization failed, rc=-1
 does not exist or your board is not connected
         this check.

avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
avrdude: ser_close(): can't reset attributes for device: Device not configured

avrdude done.  Thank you.
```

DavidEGrayson, 2019-11-25:
In the "Supported operating systems" section of your device's user's guide, we recently added a note about macOS 10.15, and a link to a page which explains the [macOS bug affecting the A-Star 32U4 Bootloader](https://www.pololu.com/docs/0J82).  The bootloader barely fits into its memory area, and if you apply that patch, it makes it even bigger.  The page I linked to above has a working bootloader you can download.

bdelapen, 2019-11-25:
Thanks you @DanGrayson !!! I don't know how I did not find that page but it did the trick for sure, I burned the bootloader and I am now able to upload sketches again. I will also cross post this reply to the original thread I referenced so that if anyone else with a Pololu board comes across that one then they can find the solution there.

Regards,

- Birger

DanGrayson, 2019-11-25:
@bdelapen -- maybe you intended to thank @DavidEGrayson instead.

closed 2019-11-25 by bdelapen