pololu/a-star #11: Warning: pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:15:6: warning: ‘void __vector_26()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]

triffid, 2022-05-18:
```
Compiling core...
"avr-g++" -c -g -Os -Wall -Wextra -std=gnu++2a -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -Wno-volatile -fno-stack-protector -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_A_STAR_328PB -DARDUINO_ARCH_AVR  -include "/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/io_328pb.h" "-I/home/triffid/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino" "-I/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb" "/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp" -o "/tmp/arduino_build_204993/core/new_vectors_328pb.cpp.o"
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:15:6: warning: ‘void __vector_26()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   15 | void __vector_26() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_26()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
"avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -fno-stack-protector -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_A_STAR_328PB -DARDUINO_ARCH_AVR  -include "/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/io_328pb.h" "-I/home/triffid/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino" "-I/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb" "/home/triffid/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino/wiring_pulse.S" -o "/tmp/arduino_build_204993/core/wiring_pulse.S.o"
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:33:6: warning: ‘void __vector_44()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   33 | void __vector_44() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_44()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:32:6: warning: ‘void __vector_43()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   32 | void __vector_43() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_43()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:31:6: warning: ‘void __vector_42()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   31 | void __vector_42() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_42()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:30:6: warning: ‘void __vector_41()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   30 | void __vector_41() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_41()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:29:6: warning: ‘void __vector_40()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   29 | void __vector_40() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_40()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:28:6: warning: ‘void __vector_39()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   28 | void __vector_39() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_39()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:27:6: warning: ‘void __vector_38()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   27 | void __vector_38() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_38()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:26:6: warning: ‘void __vector_37()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   26 | void __vector_37() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_37()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:25:6: warning: ‘void __vector_36()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   25 | void __vector_36() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_36()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:24:6: warning: ‘void __vector_35()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   24 | void __vector_35() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_35()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
[... 26 lines cut ...]
[... cut ...]
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:16:6: warning: ‘void __vector_27()’ specifies less restrictive attribute than its target ‘void __bad_interrupt_328pb()’: ‘nothrow’ [-Wmissing-attributes]
   16 | void __vector_27() __attribute__((used, weak, alias("__bad_interrupt_328pb")));
      |      ^~~~~~~~~~~
/home/triffid/.arduino15/packages/pololu-a-star/hardware/avr/4.0.2/variants/a-star328pb/new_vectors_328pb.cpp:13:6: note: ‘void __vector_27()’ target declared here
   13 | void __bad_interrupt_328pb(){ asm("jmp __bad_interrupt"); }
      |      ^~~~~~~~~~~~~~~~~~~~~
```

Seems like an easy fix

kevin-pololu, 2022-05-18:
We're having trouble reproducing your issue. What OS and Arduino version are you using, and what program are you trying to compile?

Note that those warnings are coming from some support files that were originally needed because the Arduino toolchain didn't support the ATmega328PB, but they are now unnecessary and we are looking at removing them, so this might become irrelevant soon anyway.

triffid, 2022-05-19:
I just followed https://www.pololu.com/docs/0J74/4.3 in Arduino 1.8.19 and gave the Blink example a test compile.

Running on Gentoo Linux with Arduino configured to use my system avr-gcc (12.1.0)

DavidEGrayson, 2022-05-20:
We just released version 5.0.0 of the A-Star boards package.  If you update to the latest version using the Arduino IDE's Board Manager, these warnings should go away, because the file generating them has been removed.

closed 2022-05-20 by DavidEGrayson

triffid, 2022-05-20:
Board manager only shows `4.0.2` and older versions, do I have to do something special to see the latest one?

Oh, looks like your domain is having DNS issues, that'll do it

triffid, 2022-05-20:
Working now, warnings are gone with `5.0.0` 👍