pololu/a-star #12: TWI driver broken with Version 5.0

jacky4566, 2022-08-02:
Looks like the latest changes broke some of the the TWI I2C linking.

Steps to reproduce:

- New Sketch with Polulu A-star 328PB @ 3.3V, 8 MHz
- #include <Wire.h>
- compile

Error: "'TWSR' undeclared" and other references missing.

DavidEGrayson, 2022-08-03:
Thank you for reporting this issue to us.  As a workound for now, I recommend adding the following line to the A-Star `boards.txt` to define the things that the Wire library uses:

    a-star328PB.build.extra_flags=-DTWBR=TWBR0 -DTWSR=TWSR0 -DTWAR=TWAR0 -DTWDR=TWDR0 -DTWCR=TWCR0 -DTWAMR=TWAMR0 -DTWI_vect=TWI0_vect

You can add that line anywhere in `boards.txt`.  If you are using Windows and installed the A-Star package in the usual way, that file should be in:

    C:\Users\[username]\AppData\Local\Arduino15\packages\pololu-a-star\hardware\avr\5.0.0
    
We will look into fixing this in a more permanent way.  Please let me know if you have any further questions.

DavidEGrayson, 2022-09-01:
The latest version (5.1.0) fixes the Wire library and the SPI library, and it also adds menus in the Arduino IDE so you can select which bus those libraries use.

closed 2022-09-01 by DavidEGrayson